SylvainMontagny/LoRaE5 #2: Successful test

aemmenet, 2024-03-24:
Hi Silvain

I successful tested your LoRa-E5 Arduino Library on my DIY "polylab" Development Board.

Reading and sending the values (temperature and humidity) of a DHT22 sensor works fine.
Switching LED_BUILTIN according to the PayloadDown value (0x00/0x01) also works.

The requested transmission after pressing 't' only worked after patching the library.

For my Dev. Board I had to cross out the two serial defines on top of the lora5.h file.

Since in the lorae5.cpp file at 'await NextTransmission' you use Serial.available() and Serial.read() instead of USB_Serial.avaliable() and USB_Serial.read() reading the button didn't work and i had to patch Serial.available() and Serial.read() to USB_Serial.available() and USB_Serial.read().

https://github.com/aemmenet/polylab/tree/master#lora-E5
https://github.com/aemmenet/polylab

Kind regards
             Kurt

(Switzerland near Berne)

SylvainMontagny, 2024-03-26:
Hi Kurt,
Thanks for reaching out, and thank you for your comment about the Serial. By the way, Your Polylab is a super nice board.
We are working on a new version of the library that will be available in the next few days. In will support class C and on the application side, we wanted to provide more versatility to build an application. This version won't be compatible with the v1, we would be happy to have your feedback.
Regards,
Sylvain

aemmenet, 2024-04-24:
I have successfully modified my test sketch for the V2 library. The Bosch BME280 sensor is successfully read and the data are sent as  payload. The LED_BUILTIN on my development board is turned off or on depending on the downlink payload (0x00/0x01).

closed 2024-04-24 by aemmenet

SylvainMontagny, 2024-04-25:
Nice, new improvements to configure the serial TX/RX from the application are coming soon as proposed by @SensorsIot in the Issues.