SylvainMontagny/LoRaE5 #6: raspberry pi pico on grove lora wio E5

vishnuu20, 2024-05-22:
i tested this library by connecting it to raspberry pi pico W and grove lora e5 and i couldn't upload the code since it says the board is not compatible and error appears. i would like to send sensor data through lora and can you suggest me a way to solve this problem using the grove lora e5  that i have already using now.

vishnuu20, 2024-05-22:
> Hi, I haven't tested on a rapsberry pi pico but I will as soon as can get
> one. I don't see any reason why it would work.

Thank you for the feedback. But i am really struggling with it. I tried
using grove lora E5 . Is there any compatibility issue? Once you try please
update to me . Thank you so much.

vishnuu20, 2024-05-23:
I forgot to mention i am using as923 , is that why i can't use it?

SylvainMontagny, 2024-05-23:
Can you provide the error you get ? 
Yes, the lib is not yet ready for AS923, we are going to update it soon mais unfortunately, the community will have to test it for us as we don't have an AS923 GW and are not allowed to use this band anyway.